prusa3d/Prusa-Link #958: Printer statistics are not updated when printing through Prusa Link

koos42, 2024-04-28:
Steps to repeat:
1. Check your machine printing statistics, I noticed with the `total filament` stat.
2. Print something via Prusa Link.
3. Check your stats again.

Machine runtime and total filament might be impacted.
I don't know if this is more of a Prusa Link bug, or a Prusa Firmware bug; thoughts?

```
Version
API: 0.9.0-legacy
Hostname: prusalink
Firmware: 3.13.3-7094
Server: 0.7.2
Text: PrusaLink 0.7.2
SDK: 0.7.1
Frontend: 3.12.0

System Version
Python: 3.9.2 (default, Mar 12 2021, 04:06:34) [GCC 10.2.1 20210110]
Description: Raspbian GNU/Linux 11 (bullseye)
RELEASE: 11
CODENAME: bullseye
ID: Raspbian
```

3d-gussner, 2024-05-15:
@koos42 Thanks for reporting. In the next firmware release 3.14.0 we added tracking of filament used and time of host prints to the printer statistics.
Please consider to close this issue.

3d-gussner, 2024-07-05:
@koos42 The Prusa-Firmware [3.14.0](https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0) and PrusaLink [0.8.1](https://github.com/prusa3d/Prusa-Link/releases/tag/0.8.1) have been released and should fix this issue. Please update your printer and PrusaLink to these versions and test it again.
If it is working (it should) please close this issue.