prusa3d/Prusa-Link #966: Power cut causes gcode corruption

warmekaassaus, 2024-05-24:
Setup: Mk3s+ with raspberry pi zero2W (6 of these)

Even though we have a UPS set up, occasionally power cuts still cause the printers to lose power. Subsequent prints seem to all fail at exactly the same point in the gcode file. Deleting and reuploading the same gcode file (but from a backup - so undamaged) causes the print to succeed as expected.

See below for three subsequent prints, the traces of the nozzle simply halting in place visible at exactly the same point in the print.

[image]
[image]
[image]

3d-gussner, 2024-05-24:
@warmekaassaus Does the UPS runs out of power and so causes a power loss, or does it happen all the time when power is disconnected from the UPS? Please see https://help.prusa3d.com/article/ups-backup-for-your-printer_266023

3d-gussner, 2024-07-05:
@Prusa-Support Stale issue please close it