prusa3d/Prusa-Link #970: Has support ended for the original MK3 filament sensor?

Mark-FS, 2024-05-26:
I have just installed Prusa Link on my original MK3 printer and the filament sensor will not work after the required firmware update.

If I run 3.9.3 the sensor works perfectly but Prusa Link doesn't.  If I run 3.10.1 or higher Prusa Link works but the sensor does not.

Is this a bug in the firmware or has Prusa stopped supporting the original hardware in the later firmware updates?

3d-gussner, 2024-05-27:
@Mark-FS This is a Prusa-Firmware issue we are working on and will be addressed in FW 3.14.1 see https://github.com/prusa3d/Prusa-Firmware/issues/4380

Please close this issue here.

Mark-FS, 2024-06-01:
Thanks for the reply.

FW 3.10.1 doesn't work and that was released in 2021.  Has the filament sensor been broken for that long?

Mark-FS, 2024-06-11:
@3d-gussner could I get an update on my question above (#3)?

3d-gussner, 2024-06-25:
@Mark-FS We had to rewrite some code around the filament sensor to make it work with the new MMU3 firmware, frankly this introduced this issue. The filament sensor works well except when the printer is powered on we are working on it as I pointed out above.
I guess we missed this issue as we and most users keep the filament loaded after a print and while testing it didn't happen.

Please close this issue here as it is firmware and not PrusaLink related.

3d-gussner, 2024-07-05:
@Prusa-Support Please close this issue

Mark-FS, 2025-01-20:
@3d-gussner @Prusa-Support  As you now know 3.14.1 did not solve the issue.  Do we have a new update or is the old hardware no longer supported?

3d-gussner, 2025-01-20:
@Mark-FS There is an attempt to improve the MK3 sensor see https://github.com/prusa3d/Prusa-Firmware/pull/4834