prusa3d/Prusa-Link #971: Feature request: update firmware of printer through Prusa-Link

redfast00, 2024-06-06:
I tried to print a file, and heard the printer beep. I then walked over to see what the fuss was about, and saw that there was a notification that a new firmware was available.

It would be nice if Prusa-Link could periodically fetch the latest firmware available for the connected printer, and offer to install it on the connected printer.

Printer: MK3S, connected via USB to a Raspberry Pi (so not via GPIO)

ondras12345, 2025-01-24:
I too would like to see this feature.

In the meantime, I did some research to figure out if I could perhaps do this manually via SSH:
- apparently, the firmware image has two parts and the update tool in PrusaSlicer calls `avrdude` twice: https://github.com/OctoPrint/OctoPrint-FirmwareUpdater/issues/309#issuecomment-1196465192
- it uses a prusa-specific fork of `avrdude`

I'm giving up, I don't feel like building the forked `avrdude` from source, only to find out firmware cannot be uploaded via the RPi zero GPIO port on the MK3S. (Looking at the schematic, it might be possible, but will probably require manual reset at just the right time to enter the bootloader.)

nicohervy, 2026-10-26:
We run a small fleet of MK3S+ printers, each connected by USB to a Raspberry Pi 4 running PrusaLink. Updating the printer firmware still requires moving the printer’s USB cable to a computer, despite the Pi already having the necessary USB connection.

We would like to help move this feature forward. Would the maintainers welcome a USB-only printer-firmware updater contribution, initially limited to approved official MK3S/MK3S+ releases?

Our proposed approach is a separate maintenance worker, coordinated with PrusaLink’s serial adapter: admit updates only while idle, prevent new print requests, hand over exclusive USB ownership, run the Prusa-patched flashing flow including external language data, then reconnect and verify the printer identity and firmware version. Progress should survive the browser closing or Wi-Fi disconnecting.

We have not implemented or deployed this updater yet. Before starting a contribution, could you advise whether there is existing work or a preferred extension point, and whether integration with the hosted Prusa Connect UI would require a separate backend/frontend change? We are happy to begin with the local PrusaLink maintenance API and UI.