prusa3d/Prusa-Link #972: Prusalink status isnt updating from previous state to "ready"

Zero-Circle, 2024-06-09:
When the printer is marked in prusa connect or from the menu as "ready" Prusa link doesnt reset the status of the printer to ready, it instead stays in the previous state it was in prior to being marked as "ready".

[image]
[image]

3d-gussner, 2024-06-25:
@Zero-Circle This repository is for the MK3/S/+ and MK2.5/S PrusaLink on SBC (like Raspberry Pi) please close the issue here and reopen it in https://github.com/prusa3d/Prusa-Firmware-Buddy
PrusaLink on the MINI, MK3.5, MK3.9, MK4 and XL are implemented on the controller board.

Zero-Circle, 2024-06-25:
Marked closed as this is the wrong location for this issue. Moved to this [topic here.](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4058)

closed 2024-06-25 by Zero-Circle