prusa3d/Prusa-Link #977: MMU filament change not reported

deimosfr, 2024-06-25:
Hi,

When using MMU and filament requires to be replaced, the status is not correctly reported. Here is the output I have with `curl -H "X-Api-Key: xxx" http://prusa/api/v1/status | jq` command:
``` json
{
  "job": {
    "id": 418,
    "progress": 78.00,
    "time_remaining": 840,
    "time_printing": 3390
  },
  "storage": {
    "path": "/usb/",
    "name": "usb",
    "read_only": false
  },
  "printer": {
    "state": "PRINTING",
    "temp_bed": 60.0,
    "target_bed": 60.0,
    "temp_nozzle": 220.0,
    "target_nozzle": 220.0,
    "axis_z": 30.4,
    "flow": 100,
    "speed": 100,
    "fan_hotend": 8309,
    "fan_print": 6238
  }
}
```
The reported status should be "Attention", like without MMU.

Thanks

3d-gussner, 2024-07-05:
@deimosfr Which printer and SBC do you use?
This repository is for the Prusa MK2.5/S and MK3/S/+ PrusaLink on SBC or and other Linux based system connected via RPi port or USB to the printer. See https://github.com/prusa3d/Prusa-Link/blob/master/README.md

If you have a MK4 with MMU3 please close this issue here an open it in https://github.com/prusa3d/Prusa-Firmware-Buddy as PrusaLink for 32-bit printers like MINI, MK3.5, MK3.9, MK4 and XL run on the controller board.

deimosfr, 2024-07-05:
Yes it's MMU3 and Mk4. Thanks

closed 2024-07-05 by deimosfr