prusa3d/Prusa-Link #981: PrusaLink 0.8.1 with USB OTG Wi-Fi Dongle

ingbrzy, 2024-07-03:
- Fresh install from RPi image
- PrusaLink 0.8.1
- Prusa Connect SDK 0.8.1
- MK3S+ 3.14.0-8066
- RPi Zero 2W connected via GPIO

I want to use USB OTG Wi-Fi dongle for better Wi-Fi signal reception..

When I disable RPi Wi-Fi by adding:
`dtoverlay=disable-wifi` to `/boot/firmware/config.txt`

IP address changed as expected and RPi is Online but PrusaLink is stuck on "RPi booting..." message..

How to use external Wi-Fi dongle with PrusaLink?

Thank you

ingbrzy, 2024-07-10:
Solved by commenting these lines in prusa_link.py

https://github.com/prusa3d/Prusa-Link/blob/master/prusa/link/printer_adapter/prusa_link.py#L360-L361

3d-gussner, 2024-07-10:
@ingbrzy Thanks for the update. I will ask the developers to have a look at your solution.
Can you create a branch in your fork of PrusaSlicer with the changes you made?
If that also works for the "stock" configuration with the RPi WiFi you can consider to make a PR.

ingbrzy, 2024-07-11:
@3d-gussner 
that commenting of the lines has a little disadvantage that there is not visible IP and MAC in the Network info on PrusaLink page

[image]

So it would be nice if your devs propose another solution for USB WIFI dongle on RPi Zero2W..

Thank you

ingbrzy, 2024-07-17:
@3d-gussner any response from devs?

3d-gussner, 2024-07-17:
@ingbrzy It is summer and many have vacation. As soon I have news I will let you know.

ingbrzy, 2024-08-30:
@3d-gussner 
I hope devs had a great holidays.. any response from them?

3d-gussner, 2024-08-30:
@ingbrzy Still no answer on that. I guess there are so many different USB WiFi dongles that makes it "hard" to implement.

ingbrzy, 2024-09-08:
I am using dongle TP-LINK TL-WN722N and there is no issue to get it work on Rpi OS.. it runs correctly but doesn't work for PrusaLink only due to IP update required by PrusaLink script..

3d-gussner, 2024-09-09:
@ingbrzy There might be an issue in which order things are started. The USB dongle may "load" after PrusaLink is started.