prusa3d/Prusa-Link #983: Firmware is not up-to-date, please update it! with updated firmware

slampus, 2024-07-05:
After updating printer firmware to 3.13.3 i got som errors on my Prusa Link so i decided to reinstall it. Now i cant go through the setup wizard. Found a forum post with 3 more people with the same problem. https://forum.prusa3d.com/forum/original-prusa-i3-mk3s-mk3-hardware-firmware-and-software-help/firmware-is-not-up-to-date-please-update-it-with-updated-firmware/#post-715405

[image]

Printer: MK3S+
Printer Firmware: 3.13.3

Raspberry pi w zero 2
Prusalink version: 0.8.1

3d-gussner, 2024-07-05:
@slampus Please update your printer to FW 3.14.0 https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0
There is a delay between the github release and the help.prusa3d.com/downloads website.

At the moment Prusa-Firmware and Prusa-Link are quite strict on version checking and after you updated your RPi to newest PrusaLink version 0.8.1 it expects the matching Prusa Firmware 3.14.0

As an alternative you can install the Prusa Link RPi image based on 0.7.2 if you still have it on your computer.

Sorry for the inconvenience and hope that I could help.

If that is working correctly, please close this issue.

slampus, 2024-07-06:
Thanks you @3d-gussner it worked! Did not know there where delay in release between Github and Prusa website.

closed 2024-07-06 by slampus

3d-gussner, 2024-07-08:
@slampus Glad to help and great to hear that it works.