prusa3d/Prusa-Link #986: Prusalink 0.8.1 update on Raspberry pi 2w zero

prusalubob, 2024-08-09:
Hello,

I used to have the prusa link 0.7.1 running on Raspberry pi 2w zero on my Prusa MKS3+ printer. All good I decided to update to 0.8.1 and what I did was to flash the image on the Raspberry like I first did with 0.7.1 version. The printer rebooted successfully and is asking me to access the ip adress of it in the browser to set up again the prusa link... Well... the IP is not working because my last prusa link version is still related to it thru my account, and that I can still access and I see an offline account...Now what did I missed? What should I do to access the new version of prusa link?

Thanks!

ingbrzy, 2024-08-09:
Hello..
You have to update MK3S+ Firmware to 3.14 to be able boot Prusalink 0.8.1
https://github.com/prusa3d/Prusa-Firmware/releases/tag/v3.14.0

prusalubob, 2024-08-09:
Hello,

I did it before. I have the 3.14 version. The boot run successfully and the only thing remaining is to access the 0.8.1 prusa link thru web interface, which in this point I don't know to do it....thanks.

ingbrzy, 2024-08-09:
are you able to connect to the IP which is shown on your printer screen? 
you have to complete setup wizard there..

prusalubob, 2024-08-09:
That's the issue. I cannot access the IP of the printer because related to it I think it is still the older version of Prusa link I had. And this I don't know how to override to access the new version....

ingbrzy, 2024-08-09:
try format SD card again and install fresh Prusalink 0.8.1 image..

prusalubob, 2024-08-09:
Did all over again...don't know if I have to delete somehow the 0.7.1 version related to my IP ... didn't find any info about this..

ingbrzy, 2024-08-09:
did you download full image of Prusalink 0.8.1?
https://github.com/prusa3d/Prusa-Link/releases/download/0.8.1/prusalink-0.8.1.img.xz

download it and flash via Raspberry imager to the SD card..

fwjohnson, 2024-08-10:
My MKS3 has been working with prusalink-0.7.2.img.xz.  I updated my RPI Zero 2 to prusalink-0.8.1.img.xz and MKS3 with MK3S_MK3S+_FW_3.14.0_MULTILANG.hex because I think they are co-requisite.  It boots and the LCD shows a IP addr.  If I ping that addr I get a response (if I turn the printer off I do not get a response).  Printer back on.  In my browser at 192.168.0.110/#dashboard I see the usual display and it says printer is idle.  But if I click to print a file and confirm then nothing happens.  The printer does not even start to heat.

If I back up to prusalink-0.7.2.img.xz and prusa3d_fw_MK3S_MK3S+_3_12_1_5686.hex things work again.
I have RPi imager_1.8.5.exe installed.  I did not change the configuration settings when flashing the RPi.

Also, with FW_3.14.0 the LCD item to "Load Filament" has disappeared.

prusalubob, 2024-08-11:
@fwjohnson I have installed exactly the updates you did, but when I try to access the ip the printer shows me it is not working. You did something extra before the update? Like deleting the printer you had on prusa link 0.7.2? You had to do again a wizzard set up ? Or it automatically recognized the prusa link set up?

fwjohnson, 2024-08-12:
@prusalubob To the best of my memory...
For the RPi, when I flashed prusalink-0.8.1.img.xz I did go through the config steps in the RPi imager.  When I reflashed prusalink-0.7.2.img.xz I told it to reuse the same settings.
As for the PrusaLink I had to do the wizard setup after I loaded FW_3.14.0 but I don't think I had to run it again when I reloaded FW 3.312.1.
I don't believe I did any 'delete' operation.
I did reset my local wifi router which seemed to help.
The behavior when I had 7.2.img and FW_3.14.0 loaded was that my 192.168.0.110/#dashboard gave me a Page 404 display that had the PrusaLink color scheme.

Can you do a simple command line ping to the IP addr your printer displays?