prusa3d/Prusa-Link #993: HTTPS Support

Redstonecrafter0, 2024-09-08:
I was just setting up my new MK4S and was trying to access PrusaLink with Google Chrome. I use a .dev domain locally which Chrome forces to use https as it's a secure namespace.
That got me to find out that PrusaLink currently does not support HTTPS which is a bummer.
In my case (and I assume a lot of others as well) the 3D printer is in my IoT VLAN which obviously has no internet access.
Because of that and because cloud devices can bring a lot of problems Prusa Connect is no option.

I stumbled on #769 which calls self-signed certificates difficult to manage. I absolutely disagree and would like to propose adding HTTPS support with the certificate (chain) and private key just beeing loaded from a specific path on the USB flash drive.

### TLDR
I want:
- HTTP support

Because:
- HTTPS is required for secure namespaces like .dev
- Cleartext/Base64 passwords on the wire don't do much to secure anything
- Prusa Connect is no option when in an isolated IoT VLAN without internet access
- Prusa Connect is a redundant dependency on a cloud service beeing/staying up and running

I hope for this feature to be implemented or for an answer describing in detail why HTTPS support would be unfeasable/impossible.

3d-gussner, 2024-09-18:
@Redstonecrafter0 This is the PrusaLink repository for the Prusa MK2.5/s and MK3/S/+ connected to an SBC or PC, running PrusaLink on this devices.

The MK4/S has PrusaLink on the controller board, please close this issue here and open it in the 32-bit repository https://github.com/prusa3d/Prusa-Firmware-Buddy

nerakhon, 2024-09-18:
Even though the subitter describes his issue on MK4S his still applies to PrusaLink instance on RPi connected to a printer via USB, still it is insecure to log in via HTTP. I'd say that running letsencrypt and hiding the web ui via https would still be a welcome addition, sending any password in plain text isn't something users should do in 21st century.

Redstonecrafter0, 2024-09-19:
Surely did create the issue over on the other repository: https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/4215

> Even though the submitter describes his issue on MK4S this still applies to PrusaLink instance on RPi connected to a printer via USB, still it is insecure to log in via HTTP. I'd say that running letsencrypt and hiding the web ui via https would still be a welcome addition, sending any password in plain text isn't something users should do in 21st century.

Of course it would not harm to add HTTPS support to the RPi PrusaLink version and it would be literally only three lines of code.  
But on the RPi you could also host a reverse proxy to do the TLS termination. There are many options to choose from (nginx, Apache2/httpd, traefik, caddy and many more). That way the unencrypted traffic never leaves the RPi and it stays secure. This is a common practice and serves well on the RPi.

closed 2024-09-19 by Redstonecrafter0