prusa3d/Prusa-Link #994: consider adding support for MQTT

mgx0, 2024-09-23:
Hi there, 

please consider adding support for sending MQTT messages. Such messages could contain:
- printer state
- printing status (including a picture snapshot URL)

it could be then possible to integrate into home automation system (node-red). I can imagine being notified of printing status/printer state by a way of my choice (Pushover, Telegram).

or is there a REST API to query the Prusa-link for these information already? could not find anything about it. My goal is the Prusa-link to work off the internet (offline) only connected to my home network.

thanks

3d-gussner, 2024-09-23:
@mgx0 Please have a look at https://www.home-assistant.io/integrations/prusalink/ and https://github.com/home-assistant/core/tree/dev/homeassistant/components/prusalink here you may find some ideas how to integrate PrusaLink to  your home automation system.

mgx0, 2024-09-23:
Wonderful, thanks!

closed 2024-09-23 by mgx0