prusa3d/Prusa-Link #995: Prusalink 0.81 on MK3i3 not connecting to Prua connect

AlainGeenrits, 2024-10-01:
I installed 0.81 on Rpi 3 formerly running Octoprint. The local connection and config are fine.

API
2.0.0
Hostname
prusamk3i3
Firmware
3.14.0-8066
Server
0.8.1
Text
PrusaLink 0.8.1
SDK
0.8.1
Frontend
3.12.0

However I cannot link to Prusa connect. I can login to the site fine, but get a 404 error when trying to link. This is the URL that is not loading: https://connect.prusa3d.com/add-printer/connect/1.3.0/9NKTH6DRJS?name=PrusaMK3i3&location=Hobbyroom

This is Safari on MacOS Sequoia. I locked at network in Little Sniffer and allowed a blocked connection to collector.github.com. Also tried Firefox.

felcrv, 2024-12-27:
Same issue here. I run a MK3S+ with Raspberry Pi Zero 2 and I am also running firmware 3.14.0 and PrusaLink 0.81. 
I can connect to PrusaLink perfectly fine, but as soon as I try to connect to PrusaConnect, I get an error message. I upload also pictures of the error messages. It is also weird, because I also have a second MK3S+ with the same configuration and softwares and there it works perfectly fine. 
[image]
[image]
[image]

karlg100, 2026-01-16:
I have the same problem.  I get a 404 not found from connect. it attempts to register to https://connect.prusa3d.com/add-printer/connect/1.3.1/[printer info]