prusa3d/Prusa-Link #996: Feature request; No way to remotely change filament

Escrich, 2024-10-04:
Dear all, I'm starting to use Prusa Link and Prusa connect, but I'm used to control all my other printers form my desktop computer, in the case of my Prusa, I'm not able to load or unload filament in a single operation, and I need to do it.

Just referring the filament change, even remote or locally, we really don't need to heat up the bed to change filament, sometimes we only want to move the reel to a different printer, or to a storage bin, anyway we don't need to waste energy warming the bed.

Escrich, 2024-10-04:
I hope this time the issue will not disappear as the previous one did

3d-gussner, 2024-10-07:
@Escrich Which Prusa printer are you using? This repository is for the MK2.5/S and MK3/S/+ 8-bit printers with PrusaLink running on a SBC or other Linux device.

In case you have an MINI, MK3.5/S, MK3.9/S, MK4/S or XL please close this issue ticket here and open a new one in the Buddy firmware repository https://github.com/prusa3d/Prusa-Firmware-Buddy
 
>Just referring the filament change, even remote or locally, we really don't need to heat up the bed to change filament, sometimes we only want to move the reel to a different printer, or to a storage bin, anyway we don't need to waste energy warming the bed.

On MK3S you can set in LCD menu `Settings -> HBed on load` to  `[No]`  and load and unload will not heat up the bed. Hope that helps.

Escrich, 2024-10-07:
> @Escrich Which Prusa printer are you using? This repository is for the MK2.5/S and MK3/S/+ 8-bit printers with PrusaLink running on a SBC or other Linux device.
>
> In case you have an MINI, MK3.5/S, [...]

Mu printer is an Original Prusa i3 Mk3s

Escrich, 2024-10-07:
--- On MK3S you can set in LCD menu Settings -> HBed on load to [No] and load and unload will not heat up the bed. Hope that helps.

Thanks for that, that's true, now has been done, thanks again.

Escrich, 2024-11-27:
Still no way to change remotely filament, no way for load or unload it from the web interface

Nailig, 2024-12-07:
> Still no way to change remotely filament, no way for load or unload it from the web interface

I wonder why you need it, since you need to be with the printer to physically change the filament, why add a remote option?
If you don't want to wait for heating, you can start heat remotely I guess...

Escrich, 2024-12-07:
> > Still no way to change remotely filament, no way for load or unload it from the web interface
> 
> I wonder why you need it, since you need to be with the printer to physically change the filament, why add a remote option? If you don't want to wait for heating, you can start heat remotely I guess...

Yes, you're almost right, but sometimes I need to unload several printers at same time, in my klipper ones it's too easy, I click on the unload button, and the printer heats up, and when some temperature has been reached, directly unloads the filament, then when you finish to load in the printer the new gcodes to make the new parts, you simply need to go to the printers cutting the tip in angle, changing the roll, and placing a new one, all my klipper printers are able to do the process automatically, you simply order to unload and at end take the reel away, no more intervention, I do the most seated in front of my computer.
In the case of a Prusa printer, my Mk3s, in this case, you need to go to the printer, navigate the menu to look for the unload option, push the button, choose the material loaded at machine, push the button to accept it, wait to be heated up, push the button again, (this last push-button operation must be in any case removed, has no sense at all), and finally, you take the filament away, cut the tip making an angle and place the new roll, if the printer is not near your computer, you need to stay close to the printer, since the beginning to the end of the operation, in the meanwhile you can do nothing, in the whole other cases you need to be only at end of the process, with filament almost out of the machine, the operation time is much shorter than in the Prusa case, that's the reason why I need to start the unload operation from my computer