prusa3d/Prusa-Link #1001: All data from the printer in status message

svrooij, 2024-10-30:
Currently we can only see the data below about the printer. I would like to see the following data there as well:

- `axis_y` and `axis_x`
- MMU3 filament currently in use
- MMU3 status, like `IDLE`, `PRINTING`, `CHANGING_FILAMENT`
- If it's actually moving, at that exact moment

I envision that this data could be used to automate timelapses without having to physically connect to the printer.

Once this extra data is available your could set the printer to go to a specific position after each layer, wait for 200ms. Checking for that position would allow taking timelapses with any old phone connected just to the wifi, since it will know when to take a picture by just checking the x and y position.

```json
{
    "job": {
        "id": 14,
        "progress": 67,
        "time_remaining": 3960,
        "time_printing": 8546
    },
    "storage": {
        "path": "/usb/",
        "name": "usb",
        "read_only": false
    },
    "printer": {
        "state": "PRINTING",
        "temp_bed": 60,
        "target_bed": 60,
        "temp_nozzle": 225,
        "target_nozzle": 225,
        "axis_z": 68.4,
        "flow": 100,
        "speed": 100,
        "fan_hotend": 3558,
        "fan_print": 5023
    }
}
```

TojikCZ, 2024-10-31:
Hello, unfortunately this repo concerns mainly the MK3, PrusaLink for the buddy family is located inside the firmware - Prusa-Firmware-Buddy repo. Let me know if you're ok with me transferring the issue there

svrooij, 2024-10-31:
Sure. I didn't know that. Can we find documentation somewhere what belongs where?