prusa3d/Prusa-Link #1003: Allow Customization of Time/Date Display

ryanwalder, 2024-11-06:
PrusaLink on MK4(s)

I would like to be able to modify the way PrusaLink displays Time/Date information:

Current View, uses US data format (arguably the worst!):

[image]

I would like to be able to format this however I like (ideally with the same arguments as the linux `date` command, eg: `%Y-%m-%d %H:%M:%S`).

3d-gussner, 2024-11-06:
@ryanwalder This is the PrusaLink repository for the MK2.5/S and MK3/S/+ with a SBC or Linux computer. Please close the issue here and open it in the correct repository https://github.com/prusa3d/Prusa-Firmware-Buddy

ryanwalder, 2024-11-06:
My bad! Thanks for the heads up.

closed 2024-11-06 by ryanwalder

3d-gussner, 2024-11-06:
@ryanwalder NP isn't easy to find the correct repository