prusa3d/Prusa-Link #1005: In latest versions we lost time to change value on screen

Escrich, 2024-12-06:
In the past, on Prusa MK3s screen, you could be able to see how, time to next change, and time to print end, were alternating, one after the other, lately, unfortunately, we can only see time to print end, having no idea about how long it will take until next color change, or filament change.

Same information is lost too, at Prusa Connect and Prusa Link

3d-gussner, 2024-12-13:
@Escrich This is an issue in PrusaSlicer not calculating the M73 times correctly. This broke at some point and the PrusaSlicer Team is aware of that issue. :crossed_fingers: in next PrusaSlicer Release it is working again.

3d-gussner, 2024-12-18:
@Escrich Please close this issue here as it an PrusaSlicer issue and should be fixed in PS 2.9.0-beta1

Escrich, 2024-12-18:
Ok, then looks like solved, I'll check it!

closed 2024-12-18 by Escrich