prusa3d/Prusa-Link #1006: No Lan Access

Spencer-Laframboise, 2024-12-14:
I am using the latest release of the software, 0.81, on a pi zero w plugged into via gpio to the mrk3s+. This works well for a while but at some point i just get the message no lan access. I do a fresh install without changing any settings and it works fine again.

GiorgioAresu, 2024-12-22:
I'm having the same issue with a brand new install. The same pi works just fine with octopi/octoprint, but can´t connect to wifi with prusalink

theliadir, 2025-02-05:
I'm having a similar issue. I get the ->No LAN access error.

Im using: 
- raspberry pi zero 2 w
- Prusa-Link 0.8.1
- Bondtech Prusa Firmware 3.14.0

Im using a AX1500 Router and I have the following security options: 
1. WPA2-PSK[AES]
2. WPA2-PSK[AES]+WPA2-PSK[AES]
3. WPA3-Personal
4. WPA3-Personal+WPA2-PSK[AES]
And none of those work. I tried with a different and older Router that has WPA-Person or WPA2-Personal which worked fine.

I think it might be that prusa-link does not support any of the above security options?
Is there a way to make it work?

GiorgioAresu, 2025-02-06:
@theliadir if you're using mixed WPA2/WPA3 network, I was able to make it work by adding `brcmfmac.disable_feature=0x82000` at the end of the cmdline.txt in the boot partition (see https://github.com/raspberrypi/linux/issues/4976#issuecomment-2543507101). Other distros don´t need it, the one prusalink is based on, apparently does.

duennerd, 2025-02-06:
Ah I forgot to mention that I tried that already. Still wasn't able to make it work.
But I will try again with the description in the issue you linked.

theliadir, 2025-02-07:
I was able to get it working with a new clean install.

XeNoMoRpH, 2025-06-12:
I ran into the same issue. I tried all the above plus several other things like manually modifying wpa_supplicant.conf to no avail. Once I got a mini-HDMI to HDMI cable and USB adapter for a keyboard, I ran "sudo raspi-config", System -> Wireless LAN. Entered the credentials and it was connected.

I have a Raspberry Pi Zero 2 W.

drbik, 2025-07-27:
Pi Zero 2 W – No LAN access with PrusaLink 0.8.x

Pi Zero 2 W, Original Prusa i3 MK3S/+, firmware 3.14.1

PrusaLink 0.7.0: LAN and web UI work fine

PrusaLink 0.8.1 (and 0.7.2): always shows “No LAN access” (no IP, not visible in DHCP, no web UI)

wpa_supplicant.conf is processed (disappears after boot), tested with different networks

Downgrade back to 0.7.0: LAN works again immediately, same hardware and config

Conclusion:
This is a reproducible regression in 0.8.x for Pi Zero 2 W.

gamberoillecito, 2025-07-31:
I got it working by following this suggestion:
https://github.com/raspberrypi/linux/issues/4976#issuecomment-2636137849