prusa3d/Prusa-Link #1010: Not possible to use Prusa Connect with MK3s+ & MMU2S anymore

coding-lemur, 2024-12-22:
Hey. After upgrading my MK3S+ with MMU2S to the newest firmware 3.14.1 (printer) and 3.0.3 (for the MMU2S) is not possible to set the combination as current printer in Prusa Slicer (Profile: MKS3s+ & MMU2S) and send GCODE to the printer via Prusa Connect (my Prusa Link Version 0.8.1).

Before the Firmware Upgrade it was possible to use the MK3s+ & MMu2S and MK3S+ & MMU3 Profiles

I checked the logs and found an error:

```
Dec 21 16:49:48 prusaprinter prusa.link.printer_adapter.structures.item_updater[663]: ERROR: Gather of mmu_version has failed {_gather():443} Traceback (most recent call last): File "/home/rene/.local/lib/python3.11/site-packages/prusa/link/printer_adapter/structures/item_updater.py", line 428, in _gather value = item.gather_function() ^^^^^^^^^^^^^^^^^^^^^^ File "/home/rene/.local/lib/python3.11/site-packages/prusa/link/printer_adapter/printer_polling.py", line 609, in _get_mmu_version build_match = self.do_matchable( ^^^^^^^^^^^^^^^^^^ File "/home/rene/.local/lib/python3.11/site-packages/prusa/link/printer_adapter/printer_polling.py", line 477, in do_matchable raise RuntimeError("Printer responded with something unexpected") RuntimeError: Printer responded with something unexpected
```

Hope that help for fixing that issue.

Best regards

coding-lemur

TojikCZ, 2024-12-22:
Hi, I am aware of the issue and have a fixed up version kudos @3d-gussner.
Once the christmas season is over, it is likely we'll get it out through OTA.

I can send you a fixed up link image for testing if you want. Thank you for the excellent report.

3d-gussner, 2024-12-23:
@coding-lemur Please send me an email.

coding-lemur, 2024-12-23:
I tested the new update: it resolves my problem.
Thanks at all, great work :)

closed 2024-12-23 by coding-lemur

ThomasDesloovere, 2025-01-17:
Hey i got the same problem but with a MMU3 any solutions that are public?

3d-gussner, 2025-01-17:
Here a workaround until we get the 0.8.2 released.

Search for the file `regular_expressions.py`, the other `__init__.py` is less important.
See also https://github.com/prusa3d/Prusa-Link/pull/1009/files.

The path to the file may be different, depending on the Python version.

I personally do not own a RPi, but have PrusaLink running on my PC with Ubuntu.

I can find this file in
`~/.local/lib/python3.10/site-packages/prusa/link/printer_adapter/structures`
or `/home/<my user account/.local/lib/python3.10/site-packages/prusa/link/printer_adapter/structure`

but also in `~/.local/lib/python3.9/site-packages/prusa/link/printer_adapter/structures` PL has been installed with this Python version. This one is probably an older version which I used long time ago.

Search and replace in `regluar_expressions.py`

`\*..\.$”` 

with

`\*[0-9a-f]{1,2}\.$”`

again see my PR for the needed changes.

There should be 6 parts that are changed.

I also tested if PrusaLink and PrusaConnect work with the customized `0.8.2` version in `__init__.py`.

In case you have `ssh` enabled on the RPi then you can connect to it and change it there.
As I don't own any RPis you may need to search if you can edit these files on the SD card directly.

3d-gussner, 2025-01-17:
@ThomasDesloovere Please try this above otherwise feel free to send me an email.

ThomasDesloovere, 2025-01-17:
Hi, thanks for your answer while the fix does stop it from erroring the MMU is not being detected in prusa connect and thus not in the slicer. Here are my logs:

>`Jan 17 17:27:49 McPrintFace systemd[1]: Started systemd-timesyncd.service - Network Time Synchronization.
Jan 17 17:27:49 McPrintFace systemd[1]: Reached target sysinit.target - System Initialization.
Jan 17 17:27:49 McPrintFace systemd[1]: Started systemd-tmpfiles-clean.timer - Daily Cleanup of Temporary Directories.
Jan 17 17:27:49 McPrintFace systemd[1]: Reached target time-set.target - System Time Set.
Jan 17 17:27:49 McPrintFace systemd[1]: Started apt-daily.timer - Daily apt download activities.
Jan 17 17:27:49 McPrintFace systemd[1]: Started apt-daily-upgrade.timer - Daily apt upgrade and clean activities.
Jan 17 17:27:49 McPrintFace systemd[1]: Started dpkg-db-backup.timer - Daily dpkg database backup timer.
Jan 17 17:27:49 McPrintFace systemd[1]: Started e2scrub_all.timer - Periodic ext4 Online Metadata Check for All Filesystems.
Jan 17 17:27:49 McPrintFace systemd[1]: Started fstrim.timer - Discard unused blocks once a week.
Jan 17 17:27:49 McPrintFace systemd[1]: Started logrotate.timer - Daily rotation of log files.
Jan 17 17:27:49 McPrintFace systemd[1]: Started man-db.timer - Daily man-db regeneration.
Jan 17 17:27:49 McPrintFace systemd[1]: Reached target timers.target - Timer Units.
Jan 17 17:27:49 McPrintFace systemd[1]: Listening on avahi-daemon.socket - Avahi mDNS/DNS-SD Stack Activation Socket.
Jan 17 17:27:49 McPrintFace systemd[1]: Listening on dbus.socket - D-Bus System Message Bus Socket.
Jan 17 17:27:49 McPrintFace systemd[1]: Listening on triggerhappy.socket.
Jan 17 17:27:49 McPrintFace systemd[1]: Reached target sockets.target - Socket Units.
Jan 17 17:27:49 McPrintFace systemd[1]: systemd-pcrphase-sysinit.service - TPM2 PCR Barrier (Initialization) was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/StubPcrKernelImage-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f).
Jan 17 17:27:49 McPrintFace systemd[1]: Reached target basic.target - Basic System.
Jan 17 17:27:49 McPrintFace systemd[1]: Starting avahi-daemon.service - Avahi mDNS/DNS-SD Stack...
Jan 17 17:27:49 McPrintFace systemd[1]: Started boot-message.service - Boot message.
Jan 17 17:27:49 McPrintFace systemd[1]: Started cron.service - Regular background program processing daemon.
Jan 17 17:27:49 McPrintFace systemd[1]: Starting dbus.service - D-Bus System Message Bus...
Jan 17 17:27:49 McPrintFace systemd[1]: Starting dphys-swapfile.service - dphys-swapfile - set up, mount/unmount, and delete a swap file...
Jan 17 17:27:49 McPrintFace cron[429]: (CRON) INFO (pidfile fd = 3)
Jan 17 17:27:49 McPrintFace systemd[1]: Starting e2scrub_reap.service - Remove Stale Online ext4 Metadata Check Snapshots...
Jan 17 17:27:49 McPrintFace systemd[1]: getty-static.service - getty on tty2-tty6 if dbus and logind are not available was skipped because of an unmet condition check (ConditionPathExists=!/usr/bin/dbus-daemon).
Jan 17 17:27:49 McPrintFace kernel: uart-pl011 3f201000.serial: no DMA platform data
Jan 17 17:27:49 McPrintFace systemd[1]: Starting polkit.service - Authorization Manager...
Jan 17 17:27:49 McPrintFace systemd[1]: Starting raspi-config.service - LSB: Switch to ondemand cpu governor (unless shift key is pressed)...
Jan 17 17:27:49 McPrintFace systemd[1]: Starting rpi-eeprom-update.service - Check for Raspberry Pi EEPROM updates...
Jan 17 17:27:49 McPrintFace systemd[1]: Starting sshswitch.service - Turn on SSH if /boot/ssh or /boot/firmware/ssh is present...
Jan 17 17:27:49 McPrintFace cron[429]: (CRON) INFO (Running @reboot jobs)
Jan 17 17:27:49 McPrintFace systemd[1]: Starting systemd-logind.service - User Login Management...
Jan 17 17:27:49 McPrintFace systemd[1]: systemd-pcrphase.service - TPM2 PCR Barrier (User) was skipped because of an unmet condition check (ConditionPathExists=/sys/firmware/efi/efivars/StubPcrKernelImage-4a67b082-0a4c-41cf-b6c7-440b29bb8c4f).
Jan 17 17:27:49 McPrintFace systemd[1]: Starting triggerhappy.service - triggerhappy global hotkey daemon...
Jan 17 17:27:49 McPrintFace systemd[1]: boot-message.service: Deactivated successfully.
Jan 17 17:27:49 McPrintFace avahi-daemon[426]: Found user 'avahi' (UID 104) and group 'avahi' (GID 110).
Jan 17 17:27:49 McPrintFace avahi-daemon[426]: Successfully dropped root privileges.
Jan 17 17:27:49 McPrintFace avahi-daemon[426]: avahi-daemon 0.8 starting up.
Jan 17 17:27:49 McPrintFace rpi-eeprom-update[436]: Device does not a have a Raspberry Pi bootloader EEPROM (e.g. Pi 4 or Pi 5). Skipping bootloader update.
Jan 17 17:27:49 McPrintFace systemd[1]: Finished rpi-eeprom-update.service - Check for Raspberry Pi EEPROM updates.
Jan 17 17:27:49 McPrintFace thd[447]: Found socket passed from systemd
Jan 17 17:27:49 McPrintFace systemd[1]: Started triggerhappy.service - triggerhappy global hotkey daemon.
Jan 17 17:27:49 McPrintFace systemd[1]: sshswitch.service: Deactivated successfully.
Jan 17 17:27:49 McPrintFace systemd[1]: Finished sshswitch.service - Turn on SSH if /boot/ssh or /boot/firmware/ssh is present.
Jan 17 17:27:49 McPrintFace dphys-swapfile[432]: want /var/swap=200MByte, checking existing: keeping it
Jan 17 17:27:49 McPrintFace polkitd[434]: Started polkitd version 122
Jan 17 17:27:49 McPrintFace systemd[1]: e2scrub_reap.service: Deactivated successfully.
[... cut ...]
Jan 17 17:31:37 McPrintFace poorwsgi[674]: INFO: Return file: /8657650f85be23655543d28f58941616.svg {handler_from_table():1098}
Jan 17 17:31:37 McPrintFace poorwsgi[674]: INFO: Return file: /ea93bcffb8771234f8c641f3b7e9f848.svg {handler_from_table():1098}
Jan 17 17:31:37 McPrintFace poorwsgi[674]: INFO: Return file: /ccd215a92d0427a21339a8c38e2a40f3.svg {handler_from_table():1098}
Jan 17 17:31:38 McPrintFace poorwsgi[674]: INFO: Return file: /44b654f4ca724e154a9bc60335e98847.svg {handler_from_table():1098}
`

3d-gussner, 2025-01-17:
@ThomasDesloovere please double check that the changes are still there. Please also ensure that you reboot the RPi. BTW which RPi do you use?
You may need to reconnect PrusaLink to PrusaConnect.

Please download your existing Prusa Link Setting from PrusaConnect.

Compare the `prusa_printer_seetings.ini` on the RPi with the downloaded one. Important is the `token` under `[service::connect]` and `username` and `digest` under `[service::local]`

ThomasDesloovere, 2025-01-17:
The changes are indeed changed, this is also confirmed by the printer not stating buzy in connect. I use a PI 2W, prusa link and prusa connect still have a perfect connection. On previous versions I had no problems turning on and off the mmu and switching in between.

3d-gussner, 2025-01-17:
@ThomasDesloovere Also verify that PrusaLink and PrusaConnect work when you disable the MMU in LCD -> Settings menu.

ThomasDesloovere, 2025-01-17:
Yeah they work perfectly I just printed a part without MMU file send to connect.

3d-gussner, 2025-01-17:
>On previous versions I had no problems turning on and off the mmu and switching in between.

I know with FW 3.14.0 + 3.0.2 PrusaLink was fine but with 3.14.1 + 3.0.3 PrusaLink 0.8.1 has some issues. That is why I prepared the PR. Frankly it takes some time to get it tested and released.

3d-gussner, 2025-01-17:
>Yeah they work perfectly I just printed a part without MMU file send to connect.

Please turn off the printer for a moment and turn it back on.

BTW, how did you change the 6 lines? Do you have ssh access?

ThomasDesloovere, 2025-01-17:
Yes i did't turn it off and on multiple times trying different combinations, I have indeed SSH access, as a trouble shooting step I updated raspbian so sudo apt-get update and then upgrade.

3d-gussner, 2025-01-17:
So PrusaLink and PrusaConnect work correctly with MMU disabled.

As soon you turn on the MMU you have issues with PL and PC.

Your firmware is FW 3.14.1 and MMU 3.0.3 with PrusaLink 0.8.1 + modification.

3d-gussner, 2025-01-17:
Please send me an email.