prusa3d/Prusa-Link #1018: unable to run prusalink-manager on raspberry pi

sorakiu, 2025-03-01:
Hi,
  I'm running the 0.8.1 release of the prusa connect raspberry pi image on a raspberry pi 4. I have 2 prusas to connect via USB. first one is a MK2.5S and the second is a i3 MK3s. There seems to be something wrong with the image. In the multinstance readme, it says to run prusalink-manager. 

when i do that, i get the following error:
```
$ prusalink-manager
Traceback (most recent call last):
  File "/home/sorakiu/.local/bin/prusalink-manager", line 5, in <module>
    from prusa.link.multi_instance.__main__ import main
  File "/home/sorakiu/.local/lib/python3.11/site-packages/prusa/link/multi_instance/__main__.py", line 29, in <module>
    from .controller import Controller
  File "/home/sorakiu/.local/lib/python3.11/site-packages/prusa/link/multi_instance/controller.py", line 8, in <module>
    from .ipc_queue_adapter import IPCConsumer, IPCSender
  File "/home/sorakiu/.local/lib/python3.11/site-packages/prusa/link/multi_instance/ipc_queue_adapter.py", line 8, in <module>
    from ipcqueue import posixmq  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'ipcqueue'
```

I tried installing prusalink in a venv like so:
```
python3 -m venv prusaenv
source prusaenv/bin/activate
pip install git+https://github.com/prusa3d/gcode-metadata.git
pip install git+https://github.com/prusa3d/Prusa-Connect-SDK-Printer.git
pip install git+https://github.com/prusa3d/Prusa-Link.git
sudo apt-get install libopenblas-dev
~/prusaenv/bin/prusalink-manager
```

and i get the following error
```
Traceback (most recent call last):
  File "/home/sorakiu/prusaenv/bin/prusalink-manager", line 33, in <module>
    sys.exit(load_entry_point('prusalink==0.8.1', 'console_scripts', 'prusalink-manager')())
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/bin/prusalink-manager", line 25, in importlib_load_entry_point
    return next(matches).load()
           ^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/metadata/__init__.py", line 202, in load
    module = import_module(match.group('module'))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1206, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1178, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1149, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/prusa/link/multi_instance/__main__.py", line 16, in <module>
    from ..__main__ import check_process
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/prusa/link/__main__.py", line 29, in <module>
    from .daemon import Daemon  # noqa: E402
    ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/prusa/link/daemon.py", line 12, in <module>
    from .web import WebServer, init_web_app
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/prusa/link/web/__init__.py", line 17, in <module>
    __import__('main', globals=globals(), level=1)
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/prusa/link/web/main.py", line 15, in <module>
    from pkg_resources import working_set  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 3248, in <module>
    @_call_aside
     ^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 3223, in _call_aside
    f(*args, **kwargs)
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 3261, in _initialize_master_working_set
    working_set = WorkingSet._build_master()
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 617, in _build_master
    ws.require(__requires__)
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 956, in require
    needed = self.resolve(parse_requirements(requirements))
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 815, in resolve
    dist = self._resolve_dist(
           ^^^^^^^^^^^^^^^^^^^
  File "/home/sorakiu/prusaenv/lib/python3.11/site-packages/pkg_resources/__init__.py", line 856, in _resolve_dist
    raise DistributionNotFound(req, requirers)
pkg_resources.DistributionNotFound: The 'prusa.connect.sdk.printer>=0.8.1' distribution was not found and is required by prusalink
```

so i'm kind of stumped - are there updated instructions on how to do multiple printers to a single pi or is there some step i'm missing?

TojikCZ, 2025-03-03:
Hi, the multi-instance image should work out of the box, no need to start anything. You just need to get the multi-instance variant of the image.

sorakiu, 2025-03-05:
facepalm!  thank you!