prusa3d/Prusa-Link #1023: Is it possible to containerize Prusa-Link?

xhudaman, 2025-04-23:
## I am wondering if it's possible to containerize (Dockerize) Prusa-Link?

In short, I assume the answer is yes* but I don't know the required dependencies that might turn this into a rabbit hole not worth heading down.

### Background Context
I have a personal server sitting below the desk that houses my printer and it can easily provide more power than my pi Zero W v1.1.  I am hoping that I can dockerize this and run it under the hood on my personal server.

### Questions
1. Is this a fool's errand?
2. Are there any gotchas that I need to look out for?
3. From the looks, it seems as though PrusaLink is a python-based daemon/server running on a raspbian/debian based OS.  Is that correct?

If not, I likely won't maintain a public version "officially" but will happily open source the resulting image/repo should I succeed.

Thanks in advance for the help!

3d-gussner, 2025-04-23:
@xhudaman

>1. Is this a fool's errand?

No it is a valid / good idea.

>2. Are there any gotchas that I need to look out for?

At the moment my multi instance isn't working well, to be honest didn't try hard to get it running.
- I forked and then cloned my Prusalink repository fork on my Ubuntu PC
  - Advantage of a fork is that you can change things. Provide PRs if you like. Or merge other PRs from the community.
  - `pip install .` builds PrusaLink

I haven't tried yet to run it as an service as I switch for testing quite often between different solutions or serial terminal programs.
 

>3. From the looks, it seems as though PrusaLink is a python-based daemon/server running on a raspbian/debian based OS. Is that correct?

Yes PrusaLink is python based BUT not dependent on PRi Debian OS. I have it running on Ubuntu for example. It works on Denian and probably on other Linux.

Ensure that you used a good USB cable when connecting the server with your printer. USB cables are often an issue.
Flashing the printer still needs to be done via PrusaSlicer.

3d-gussner, 2025-04-23:
@xhudaman Which server OS do you run?

I think if you get it working the community would appreciate to know how it can be done. So please feel free to share your results.

xhudaman, 2025-04-23:
Thanks for the quick reply @3d-gussner!

My server is actually a full Ubuntu 24.04 Desktop installation. It's nothing fancy, just my first home server that I wanna use for a few fun projects around the house.

If it's just a python server I think I can get it running really fast and easily.    If it can run under Ubuntu then I'm certain I can run it on a slim or maybe alpine linux image.

I'll grab a fork and build from source in that case.  Gives me flexibility with the docker build so that I can keep the build down to a cached process and add it to any image base layer.

I'll report back here if I make progress and I'll include a github repo & dockerhub url.  It likely won't be tonight as it's late and I'm recovering from a flu, but def gonna try to put some time into it.  It would be awesome if I could reclaim the pi for another project!

xhudaman, 2025-04-24:
Ok, so this is gonna be a whole brain dump so pardon me for that.

I took a stab at it today and managed to both get the image built and running in a local container on my personal laptop!

## Starting solution

> [!NOTE]
> Turns out the README here had quite some good info about building it on other distros.  Alpine Linux didn't work for me, but it may just be unfamiliarity with alternative deps used in APK.

For the time being, I've used the following setup:
**base image**: `debian:stable-slim`
**PrusaLink repo**: Cloned into container from git using `https`, may change strategy as the image gets more thoughtfully built
**pip install command**: `pip install .`
**PrusaLink.ini**: default with slight modifications to default username

**Dockerfile**:
  ```dockerfile
    FROM debian:stable-slim AS build

    WORKDIR /prusalink

    RUN apt update

    RUN apt install -y \
      git \
      python3-pip \
      libcap-dev \
      libmagic1 \
      libturbojpeg0 \
      libatlas-base-dev \
      python3-numpy \
      libffi-dev \
      libopenblas0

    ENV PIP_BREAK_SYSTEM_PACKAGES=1

    RUN git clone https://github.com/prusa3d/Prusa-Link.git .

    RUN pip install .

    COPY prusalink.ini /etc/prusalink/prusalink.ini

    FROM build AS run

    RUN useradd --user-group -m --shell /bin/bash prusalink

    USER prusalink

    EXPOSE 80 # not sure if this one is needed put it here for funsies since I saw the port was used
    EXPOSE 8080

    CMD ["prusalink", "-c", "/etc/prusalink/prusalink.ini", "-f"]
  ```
### Pics or it didn't happen (obligatory)
If you look closely at my address bar you can see this is in fact running on my machine's `localhost`.
[image]

### Needs testing
- Forward COM device from host machine to container to control printer over USB
- Multi-instance setup (pretty low priority until we know other stuff works)
- Can we get it on Alpine Linux for super tiny image sizes?  Debian isn't the worst here either, but I'm a nut about useless image size 😜 

To end things off I'll throw this in a gist or a repo online once I get further testing done.  Not sure if I'll do it in a fork of this repo or a separate one since it really can be completely separated out.  I'd be down to hear arguments for which way to go on that!

xhudaman, 2025-04-26:
@3d-gussner I want to build this image in a multi-stage build, where the first stage is for building PrusaLink and the latter just for running.  After running `pip install .` what files would I need to copy to the new layer to run the resulting `prusalink` binary?  Would it just be the binary itself or are there other deps I'd need to copy?

I've got my home server now running Proxmox, figured out the setup needed to have usb passthrouh work correctly, and I wanna test setting up the container with my printer attached.  The above questions are to help me publish an image on a temp dockerhub repo to test.

xhudaman, 2025-04-26:
Very rudimentary setup but **there** it is!!!

[image]

Running using the image [at `xhudaman/prusa-link-docker` ](https://hub.docker.com/r/xhudaman/prusa-link-docker) gonna test uploading a print file and running it

xhudaman, 2025-07-29:
> [@xhudaman](https://github.com/xhudaman) could you provide a minimal working example how to run your container? When I try to run it via `docker run xhudaman/prusa-link-docker`, it gets created and immediately exits with the following "State" obtained by `docker inspect`:
> 
> ```
> "State": {
>             "Status": "exited",
>             "Running": false,
>             "Paused": false,
>             "Restarting": false,
>             "OOMKilled": false,
>             "Dead": false,
>             "Pid": 0,
>             "ExitCode": 0,
>             "Error": "",
>             "StartedAt": "2025-07-28T16:22:29.539247294Z",
>             "FinishedAt": "2025-07-28T16:22:32.47560745Z"
>         },
> ```

It's been a while since I got this setup and I've been running it in a server that spins it up automatically. 

On the surface my best guess is that it may be missing some config that you need in order to get this running for yourself.  If I'm correct you'll need to pull the repo and then update the config file with the correct settings before building the final image and spinning it up.

I'll get the repo pushed to remote.  From there, give it a go and let me know how it works.  Happy to provide some more direct help through discord (same username) if you need as well.